Testato/SoftwareWire #3: ARM enviroment support?

HansHint, 2016-09-20:
can anybody tell me what exactly should be changed to get this working on an ARM system (Arduino DUE?)

Koepel, 2016-09-20:
The macros i2c_sda_lo() and the others can be replaced by digitalWrite(), pinMode() and digitalRead(). The macros are for AVR chips to get the maximum speed.

The OneWire library is currently maintained by Paul Stoffregen, and it contains high speed macros for many processors, including the Due: [https://github.com/PaulStoffregen/OneWire/blob/master/OneWire.h](url)
Those could be used for this software library.

bperrybap, 2016-10-30:
Actually, you can use indirect port i/o like what is in the i2c_xxx_hi/lo macros on pretty much any core since they all will have the underlying supporting macros to get the port address and bit mask. The trick is that the width of the i/o register is not always 8 bits. So the class declarations for the masks, port, pin, and dir registers etct needs to be smarter to ensure that the proper size/width is used in the declaration. 
i.e. on AVR the width is 8 bits, but on some other processors like ARM and PIC32 it is typically 32 bits.
I have done this kind of stuff in other libraries.

Another alternative is to detect certain conditions that you know how to support, do the apprpriate thing, but if it is being compiled on other targets have a fall back mode.

The bigger concern (and this needs a separate issue that I'm about to create) is that the i2c_xxx_hi/lo macros are not atomic and therefore not interrupt safe.
So if the pins are shared with a library that alters the register at ISR lever, this library will corrupt the register.

closed 2018-04-24 by Testato