Testato/SoftwareWire #4: library lincense?

bperrybap, 2016-10-28:
There is no clear license for this library.
From the comments, it appears to be a derivative of this library: 
https://github.com/todbot/SoftI2CMaster
which is licensed as CC BY-SA 3.0
But that license is not really compatible with LGPL and GPL code.
Read more about the licensing in an issue I posted in SoftI2CMaster:
https://github.com/todbot/SoftI2CMaster/issues/14

bperrybap, 2016-10-28:
The reason I bring this up is that I have an Arduino library (hd44780) that  is licensed GPL v3 and would like users of that library to be able to use it with SoftwareWire.
However, from my understanding of CC BY-SA,  it appears that CC BY-SA 3.0 is only compatible with itself which means it can't be linked with other LGPL or GPL code.
While this is not an issue for hobby projects since a person is free to mix and match anything for their personal use, it would be an issue for anyone that decides to distribute their code as part of a larger project especially if that were a commercial product.

This same issue will exist for many other Arduino libraries.

Koepel, 2016-10-28:
This is build on the SoftI2CMaster before it had a licence.
I prefer it to be Public Domain.
What would you suggest ?

bperrybap, 2016-10-28:
Given that this library is a derivative of SoftI2CMaster are you sure that it can be moved to Public Domain? Just because it didn't specify a license doesn't mean that it could be used/re-used with any license or for any purpose - the original copyright holders would have to agree.

But in terms of licensing, It depends on what you want.
Public Domain makes the code FreeWare which means anybody can use it and modify it any way they want including using it in a commercial closed source product that they sell.
They also would never have to provide back any sort of modifications/updates/improvements/enhancements.

On the other end of the spectrum is GPL v3, which mandates that the code can only be used in open source projects/products so that the code can never be used in a closed source projects/product.
This is my preference for open source projects that I am involved with. The reason being is that I only want to expend my time, resources, and share code with those that also create open source projects/products. So I don't want my code used in closed source projects/products. Otherwise, I would be an unpaid employee of the entity using the code in a closed source product.

LGPL is a little less restrictive than GPL but still mandates that any updates be provided back to the original project.

There are many licenses out there with varying degrees of "free-ness". 
The BSD license is probably one of the closest licenses to freeware.

From my understanding, CC BY-SA is one of the worst since it is really only intended for a entire blob vs something like a s/w library which is only 1 piece of a larger final image that is created.
i.e. CC BY-SA would be good for a document, or a h/w design where it is a complete work.

Testato, 2016-10-28:
If I remember correctly, a code released without licence automatically become a public domain code. Do you know something's about this sentence ?

If it is true, we can change the licence to everything.

In this case i prefer last GPL, for your same reason.

For curiosity, if we set the licence to last GPL may we in the future change it ?

Koepel, 2016-10-28:
Testato, that is GPLv3 ?
I still prefer Public Domain or even WTFPL, and I'm happy with LGPL or BSD.

bperrybap, 2016-10-28:
I think code released without a license is normally owned by the original author(s) with no automatic rights to use granted to anyone else, at least in the USA. As anything written is automatically given copyright status to the original authors.
The original author can choose his license or how/if he wants to allow others to use it.

Changing licenses can be tricky. Code can always be re-licensed by the authors but all the authors have to agree. It depends on the specific licensed used and which version. LGPL 2.1 had a default clause that allows anyone to fork and convert it to any other newer LGPL or GPL license. GPL v3 has the same. The "or newer" clause can be removed by the original authors. But once code is converted from LGPL to GPL, by someone other than the authors who create their own fork, then that specific version/fork can not be converted back to LGPL even by the original authors. And no code in that GPL fork can be moved back into the original LGPL code.

The original authors can modify their own code and change licenses, but they couldn't say pull in any updates in a fork of the code that had been converted GPL back into their LGPL project.
i.e. code can move from LGPL to GPL, but not the other way.

Also, if the original authors changed the license, they can't revoke any previous licenses on older versions of the code.

Testato, 2016-10-28:
We should decide what we want,
examples if someone add some function or bug correction to this library, i prefer see rhis change and learn from it, add it to this lib, etc.
Only gplv3 can guarantees this, instead he can use this lib in a closed project whitout contribute to this

Koepel, 2016-10-28:
Then GPLv3 it is.
I did not use any code from the Arduino Wire library, only the documentation to be compatible. That should be no problem I assume.

Testato, 2016-10-28:
Also if you/we used some arduino Wire piece of code, it is possible set the licence to GPLv3, because the Arduino libs are under a LGPL licence.

So if I understand you agree to set the GPLv3 for our SoftwareWire ?

Koepel, 2016-10-29:
Yes.

Testato, 2016-10-29:
Added licence file.
Thanks @bperrybap for this discussion

https://github.com/Testato/SoftwareWire/blob/master/Licence.md

closed 2016-10-29 by Testato

bperrybap, 2016-10-29:
That is great. Thanks guys.