Testato/SoftwareWire #6: _timeout needs to be unsigned

bperrybap, 2017-01-15:
The _timeout variable needs to be **unsigned long** rather than **long** which is signed.

Koepel, 2017-01-15:
Not really. It is better and nicer if it is an unsigned long, but it does not need to be unsigned long.

The substraction with millis() en previousMillis has to be with done with unsigned long. But the resulting compare can be done to a long or a integer as well.
The 'long' means that the timeout for the Slave stretching the clock pulse is 25 days instead of 50 days. Since a normal number would be 100ms, I don't worry about it.
The goal is to be compatible with the Arduino libraries, and they have a 'long' : https://www.arduino.cc/en/Serial/setTimeout , https://www.arduino.cc/en/Reference/StreamSetTimeout

If you don't mind, I keep the 'long' for now.

bperrybap, 2017-01-15:
The overall goal is to eliminate the warning created by comparing signed vs unsigned values in the SoftwareWire.cpp code.
You can do that by either
- changing _timeout to an unsigned value
(There is no need to change the type of the parameter used in setTimeout() to unsigned long)
- adding an explicit cast to all the lines that use _timeout in comparisons to eliminate the warning.

Koepel, 2017-01-15:
I like that. I have applied two small fixes. The internal _timeout is now unsigned long, but the parameter timeout is still 'long', and a cast is added when copying to the internal _timeout.
Thank you.

closed 2018-04-24 by Testato