Testato/SoftwareWire #10: Error compiling for NodeMCU in Arduino environment

AlexZibin, 2018-04-02:
Hi, 

The Arduino compiler reports the following error:
 ``` 
Documents\Arduino\libraries\SoftwareWire\SoftwareWire.cpp:146:16: error: cannot convert 'volatile uint32_t* {aka volatile unsigned int*}' to 'volatile uint8_t* {aka volatile unsigned char*}' in assignment
   _sdaPortReg  = portOutputRegister(port);
                ^
Using library SoftwareWire at version 1.4.1 in folder: Documents\Arduino\libraries\SoftwareWire 
Using library Rtc_by_Makuna at version 2.0.2 in folder: Documents\Arduino\libraries\Rtc_by_Makuna 
exit status 1
Error compiling for board NodeMCU 1.0 (ESP-12E Module).
```

Compiling for Arduino Nano goes correct without any errors.

 #### Please help to compile for NodeMCU without errors, thank you in advance!

Koepel, 2018-04-02:
Thanks, that is indeed not implemented. The registers and bits are stored as 8-bit variables for the AVR family microcontrollers. The ESP8266 has registers of 32 bits.

The ESP2866 has a software Wire library. Why did you change to this one ?

The most compatible library is the [SlowSoftWire](https://github.com/felias-fogg/SlowSoftWire). That is a wrapper around another library, so you need both. The libraries use only normal digitalWrite() and digitalRead().

AlexZibin, 2018-04-02:
Thank you for prompt answer!

closed 2018-04-24 by Testato