Testato/SoftwareWire #18: Linker error for Arduino Uno Wifi Rev.2

jgromes, 2019-01-28:
Hi,

There's a linker error in the library: when compiling examples for Arduino Uno Wifi Rev.2 (with Arduino.h patched from #17), linker returns the following error:

```
C:\Users\<username>\AppData\Local\Temp\<tempfile>.ltrans0.ltrans.o:(.rodata+0x38): undefined reference to `SoftwareWire::begin(unsigned char)'

collect2.exe: error: ld returned 1 exit status
```

This can be resolved by adding dummy begin methods into implemenation file:
```
void SoftwareWire::begin(uint8_t addr) { }
void SoftwareWire::begin(int addr) { }
```

I'm not sure why this only happens to Arduino Uno Wifi and not other boards.

facchinm, 2019-01-28:
@jgromes does the library work as intended after applying the patch from #17 ? If so, would you mind  posting a PR on https://github.com/arduino/ArduinoCore-megaavr so we can close https://github.com/arduino/ArduinoCore-megaavr/issues/14? Thanks!

jgromes, 2019-01-28:
@facchinm I don't have the WiFi board, so I can't check it by myself. I'll open a PR once it's verified and working.

Testato, 2021-07-02:
Fixed https://github.com/arduino/ArduinoCore-megaavr/pull/15

closed 2021-07-02 by Testato