Testato/SoftwareWire #21: Problem with servo's

wjbraat, 2019-03-08:


wjbraat, 2019-03-08:
I am using an Arduino Pro Mini on 5V....

Koepel, 2019-03-08:
Yes, it does interfere.
Thank you for reporting this issue and for your clear explanation.

It is the same issue as issue https://github.com/Testato/SoftwareWire/issues/5

The Servo library uses one timer with one interrupt. In the interrupt routine the PWM signals are written to the pins. To do that, the interrupt must run at a high rate. The SoftwareWire library reads a register, changes a bit, and writes the register back. That is not safe.

The [OneWire/OneWire.cpp](https://github.com/PaulStoffregen/OneWire/blob/master/OneWire.cpp) library uses `noInterrupt()` and `interrupt()` every time a register is changed. We have to add that as well.

Can you use a different port to avoid it?

wjbraat, 2019-03-08:
Thanks you for the quick response. 
When I moved the software I2C to PortC (A0-SDA and A1-SCL) the problem was solved!

Regards, Willem

wjbraat, 2019-03-08:
It is not often that one uses all 6 PWM ports. For this design it was necessary.
So, the workaround is quite simple: don't run SoftwareWire on a port that drives a servo... :-)
Willem