Testato/SoftwareWire #22: Conflict with the normal "wire"?

lukeskymuh, 2019-04-27:
I tried to use the Softwarewire library with the normal wire library in the same scetch to comunicate with a master(Mindstorms EV3) and a slave(I2C sensor). Each function for itself works but when I call  "myWire.requestFrom(MPU_ADDR, 14, true);" the normal wire returns  bougus results to the Mindstorms when it requests "sendData" is called. If I uncomment this line the numbers 1 to 8 apear as you would expect. Any adivice?

```
#include <SoftwareWire.h>
#include <Wire.h>

SoftwareWire myWire( 7, 8);

#define SLAVE_ADDRESS 0x04
const int MPU_ADDR = 0x68; // I2C address of the MPU-6050. If AD0 pin is set to HIGH, the I2C address will be 0x69.

void setup() 

{

    Serial.begin(9600);         // start serial for output
    Wire.begin(SLAVE_ADDRESS);
    Wire.onReceive(receiveData);
    Wire.onRequest(sendData);
    Serial.println("Ready!");

    myWire.begin();
    myWire.beginTransmission(MPU_ADDR); // Begins a transmission to the I2C slave (GY-521 board)
    myWire.write(0x6B); // PWR_MGMT_1 register
    byte zero=0;
    myWire.write(zero); // set to zero (wakes up the MPU-6050)
    myWire.endTransmission(true);

}

int val,flag=0,index=0;

uint8_t c[]={1, 2, 3, 4, 5, 6, 7, 8};

char buf[8];

void loop()

{

}

void receiveData(int byteCount)

{

    while(Wire.available()>0) 
    {
      val=Wire.read();
      Serial.print((char)val);
      flag=1;
    }    
    myWire.requestFrom(MPU_ADDR, 14, true); // request a total of 7*2=14 registers
}

// callback for sending data

void sendData()
{
  uint8_t c[]={1, 2, 3, 4, 5, 6, 7, 8};
  Wire.write(c,8);  
  Serial.println(c[7]);
}
```

Koepel, 2019-04-27:
It might take too long. The request handler should respond as fast as possible. Any code that stops interrupts or interrupt routines that take a long time influences the reaction time for the request handler. 
Using the Serial functions in a interrupt routine should also be avoided. 
Can you move the usage of SoftwareWire and Serial into the Arduino loop() ?

lukeskymuh, 2019-04-28:
I had removed all Serial communication and between the reciveData and SendData is a 0.5 s delay. No sucess.

Koepel, 2019-04-28:
Did you also move the myWire.requestFrom() to the Arduino loop() ?
It might be the same bug as: https://github.com/Testato/SoftwareWire/issues/19#issuecomment-472632760