Testato/SoftwareWire #31: Unable to compile for Heltec Wifi Kit 32

misaalanshori, 2021-06-04:
I'm trying to use SoftwareWire on a Heltec Wifi Kit 32 (ESP32 board) and I keep getting a compiling error

Below is the code I'm trying to use to test. its the i2c_scanner but i replaced Wire with the SoftwareWire
Code:
```
#include <SoftwareWire.h>

// --------------------------------------
// i2c_scanner

SoftwareWire SWire(5, 14);
 
void setup()
{
  SWire.begin();
 
  Serial.begin(9600);
  while (!Serial);             // Leonardo: wait for serial monitor
  Serial.println("\nI2C Scanner");
}
 
 
void loop()
{
  byte error, address;
  int nDevices;
 
  Serial.println("Scanning...");
 
  nDevices = 0;
  for(address = 1; address < 127; address++ )
  {
    // The i2c_scanner uses the return value of
    // the Write.endTransmisstion to see if
    // a device did acknowledge to the address.
    SWire.beginTransmission(address);
    error = SWire.endTransmission();
 
    if (error == 0)
    {
      Serial.print("I2C device found at address 0x");
      if (address<16)
        Serial.print("0");
      Serial.print(address,HEX);
      Serial.println("  !");
 
      nDevices++;
    }
    else if (error==4)
    {
      Serial.print("Unknown error at address 0x");
      if (address<16)
        Serial.print("0");
      Serial.println(address,HEX);
    }    
  }
  if (nDevices == 0)
    Serial.println("No I2C devices found\n");
  else
    Serial.println("done\n");
 
  delay(5000);           // wait 5 seconds for next scan
}
```

and this is the error i keep getting:
Error:
```
Arduino: 1.8.15 (Linux), Board: "Heltec WiFi Kit 32, Disabled, 240MHz (WiFi/BT), 921600, None"

/home/adminispwd/Arduino/libraries/SoftwareWire/SoftwareWire.cpp: In constructor 'SoftwareWire::SoftwareWire()':
/home/adminispwd/Arduino/libraries/SoftwareWire/SoftwareWire.cpp:109:28: error: no matching function for call to 'TwoWire::TwoWire()'
 SoftwareWire::SoftwareWire() 
                            ^
In file included from /home/adminispwd/Arduino/libraries/SoftwareWire/SoftwareWire.h:6:0,
                 from /home/adminispwd/Arduino/libraries/SoftwareWire/SoftwareWire.cpp:63:
/home/adminispwd/.arduino15/packages/esp32/hardware/esp32/1.0.6/libraries/Wire/src/Wire.h:68:5: note: candidate: TwoWire::TwoWire(uint8_t)
     TwoWire(uint8_t bus_num);
     ^
/home/adminispwd/.arduino15/packages/esp32/hardware/esp32/1.0.6/libraries/Wire/src/Wire.h:68:5: note:   candidate expects 1 argument, 0 provided
/home/adminispwd/.arduino15/packages/esp32/hardware/esp32/1.0.6/libraries/Wire/src/Wire.h:38:7: note: candidate: constexpr TwoWire::TwoWire(const TwoWire&)
 class TwoWire: public Stream
       ^
/home/adminispwd/.arduino15/packages/esp32/hardware/esp32/1.0.6/libraries/Wire/src/Wire.h:38:7: note:   candidate expects 1 argument, 0 provided
/home/adminispwd/Arduino/libraries/SoftwareWire/SoftwareWire.cpp: In constructor 'SoftwareWire::SoftwareWire(uint8_t, uint8_t, boolean, boolean)':
/home/adminispwd/Arduino/libraries/SoftwareWire/SoftwareWire.cpp:113:103: error: no matching function for call to 'TwoWire::TwoWire()'
 SoftwareWire::SoftwareWire(uint8_t sdaPin, uint8_t sclPin, boolean pullups, boolean detectClockStretch)
                                                                                                       ^
In file included from /home/adminispwd/Arduino/libraries/SoftwareWire/SoftwareWire.h:6:0,
                 from /home/adminispwd/Arduino/libraries/SoftwareWire/SoftwareWire.cpp:63:
/home/adminispwd/.arduino15/packages/esp32/hardware/esp32/1.0.6/libraries/Wire/src/Wire.h:68:5: note: candidate: TwoWire::TwoWire(uint8_t)
     TwoWire(uint8_t bus_num);
     ^
/home/adminispwd/.arduino15/packages/esp32/hardware/esp32/1.0.6/libraries/Wire/src/Wire.h:68:5: note:   candidate expects 1 argument, 0 provided
/home/adminispwd/.arduino15/packages/esp32/hardware/esp32/1.0.6/libraries/Wire/src/Wire.h:38:7: note: candidate: constexpr TwoWire::TwoWire(const TwoWire&)
 class TwoWire: public Stream
       ^
/home/adminispwd/.arduino15/packages/esp32/hardware/esp32/1.0.6/libraries/Wire/src/Wire.h:38:7: note:   candidate expects 1 argument, 0 provided
/home/adminispwd/Arduino/libraries/SoftwareWire/SoftwareWire.cpp:132:16: error: cannot convert 'volatile uint32_t* {aka volatile unsigned int*}' to 'volatile uint8_t* {aka volatile unsigned char*}' in assignment
   _sdaPortReg  = portOutputRegister(port);
                ^
/home/adminispwd/Arduino/libraries/SoftwareWire/SoftwareWire.cpp:133:16: error: cannot convert 'volatile uint32_t* {aka volatile unsigned int*}' to 'volatile uint8_t* {aka volatile unsigned char*}' in assignment
   _sdaDirReg   = portModeRegister(port);
                ^
/home/adminispwd/Arduino/libraries/SoftwareWire/SoftwareWire.cpp:134:16: error: cannot convert 'volatile uint32_t* {aka volatile unsigned int*}' to 'volatile uint8_t* {aka volatile unsigned char*}' in assignment
   _sdaPinReg   = portInputRegister(port);      // PinReg is the input register, not the Arduino pin.
                ^
/home/adminispwd/Arduino/libraries/SoftwareWire/SoftwareWire.cpp:138:16: error: cannot convert 'volatile uint32_t* {aka volatile unsigned int*}' to 'volatile uint8_t* {aka volatile unsigned char*}' in assignment
   _sclPortReg  = portOutputRegister(port);
                ^
/home/adminispwd/Arduino/libraries/SoftwareWire/SoftwareWire.cpp:139:16: error: cannot convert 'volatile uint32_t* {aka volatile unsigned int*}' to 'volatile uint8_t* {aka volatile unsigned char*}' in assignment
   _sclDirReg   = portModeRegister(port);
                ^
/home/adminispwd/Arduino/libraries/SoftwareWire/SoftwareWire.cpp:140:16: error: cannot convert 'volatile uint32_t* {aka volatile unsigned int*}' to 'volatile uint8_t* {aka volatile unsigned char*}' in assignment
   _sclPinReg   = portInputRegister(port);
                ^
/home/adminispwd/Arduino/libraries/SoftwareWire/SoftwareWire.cpp: In member function 'void SoftwareWire::printStatus(Print&)':
/home/adminispwd/Arduino/libraries/SoftwareWire/SoftwareWire.cpp:533:27: error: cast from 'volatile uint8_t* {aka volatile unsigned char*}' to 'uint16_t {aka short unsigned int}' loses precision [-fpermissive]
   Ser.println( (uint16_t) _sdaPortReg, HEX);
                           ^
/home/adminispwd/Arduino/libraries/SoftwareWire/SoftwareWire.cpp:535:27: error: cast from 'volatile uint8_t* {aka volatile unsigned char*}' to 'uint16_t {aka short unsigned int}' loses precision [-fpermissive]
   Ser.println( (uint16_t) _sclPortReg, HEX);
                           ^
/home/adminispwd/Arduino/libraries/SoftwareWire/SoftwareWire.cpp:537:27: error: cast from 'volatile uint8_t* {aka volatile unsigned char*}' to 'uint16_t {aka short unsigned int}' loses precision [-fpermissive]
   Ser.println( (uint16_t) _sdaDirReg, HEX);
                           ^
/home/adminispwd/Arduino/libraries/SoftwareWire/SoftwareWire.cpp:539:27: error: cast from 'volatile uint8_t* {aka volatile unsigned char*}' to 'uint16_t {aka short unsigned int}' loses precision [-fpermissive]
   Ser.println( (uint16_t) _sclDirReg, HEX);
                           ^
/home/adminispwd/Arduino/libraries/SoftwareWire/SoftwareWire.cpp:541:27: error: cast from 'volatile uint8_t* {aka volatile unsigned char*}' to 'uint16_t {aka short unsigned int}' loses precision [-fpermissive]
   Ser.println( (uint16_t) _sdaPinReg, HEX);
                           ^
/home/adminispwd/Arduino/libraries/SoftwareWire/SoftwareWire.cpp:543:27: error: cast from 'volatile uint8_t* {aka volatile unsigned char*}' to 'uint16_t {aka short unsigned int}' loses precision [-fpermissive]
   Ser.println( (uint16_t) _sclPinReg, HEX);
                           ^
exit status 1
Error compiling for board Heltec WiFi Kit 32.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

```

I tried my best to understand the error but I'm still very new to Arduino and C++ so I don't really understand it.  The error also happens with the Small_example.ino example code

is this an incompatibility with the esp32?

Testato, 2021-06-04:
The library is, for now, only compatible with AVR Mcu (atmega328 etc).
It can be modified for work on other architecture, like ESP, but nobody for now are working on it.
Here an example https://github.com/Testato/SoftwareWire/issues/26

PullRequest are welcom

misaalanshori, 2021-06-05:
Hmm okay, I'm sorry i can't contribute. I do not understand enough of this.